simpleble/simpleble #222: Device with MTU zero and Service_Count zero in Linux but not Windows

GrahamNZ, 2023-03-12:
Most devices work fine but I have one that reports MTU as zero and Service Count as zero in Linux, and so is unable to be used. In Windows it works fine. It makes no difference if the device is paired or not (as long as the paired connection isn't active of course). The device is different to others as it allows USB connections (via /dev/ttyUSB0) and BT connections (via /dev/rfcomm0) as well as BLE but I don't know if that is a factor.

I'm looking at the device using InsideBlue, SimpleBleConnectExample and my own program.  All of these are using Erik's Pascal bindings, but I have no knowledge of how to compile the original examples in Linux so can't test with those.

GrahamNZ, 2023-03-13:
OK, found another problem device. I compiled the examples in both Win64 and Linux Mint 21 (64 bit) and ran the connect program twice on each, with two different devices. These are the SimpleBLE examples, not the Pascal binding ones.

In Windows everything is correct (as I know from using other BLE software):
```
D:\SimpleBLE-0.6.0\build_simpleble_examples\bin\Release>example_connect
[INFO] SimpleBLE: D:\SimpleBLE-0.6.0\simpleble\src\backends\windows\Utils.cpp:33 in initialize_winrt: CoGetApartmentType: cotype=-1, qualifier=0, result=800401F0
[INFO] SimpleBLE: D:\SimpleBLE-0.6.0\simpleble\src\backends\windows\Utils.cpp:41 in initialize_winrt: RoInitialize: result=0
Using adapter: ACERASPIRE - Top [58:96:1d:d1:53:16]
Scan started.
Scan stopped.
The following devices were found:
[0] DEVICE XXXXX [34:81:f4:eb:35:53]
Please select a device to connect to (0-0): 0
Connecting to DEVICE XXXXX [34:81:f4:eb:35:53]
Successfully connected.
MTU: 157
Service: 00001800-0000-1000-8000-00805f9b34fb
  Characteristic: 00002a00-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a01-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a04-0000-1000-8000-00805f9b34fb
    Capabilities: read
Service: 0000180a-0000-1000-8000-00805f9b34fb
  Characteristic: 00002a23-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a24-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a25-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a26-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a27-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a28-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a29-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a2a-0000-1000-8000-00805f9b34fb
    Capabilities: read
Service: 49535343-fe7d-4ae5-8fa9-9fafd205e455
  Characteristic: 49535343-026e-3a9b-954c-97daef17e26e
    Capabilities: write_request notify
    Descriptor: 00002902-0000-1000-8000-00805f9b34fb
  Characteristic: 49535343-1e4d-4bd9-ba61-23c647249616
    Capabilities: read write_request write_command notify
    Descriptor: 00002902-0000-1000-8000-00805f9b34fb
  Characteristic: 49535343-6daa-4d02-abf6-19569aca69fe
    Capabilities: read write_request
  Characteristic: 49535343-8841-43f4-a8d4-ecbe34729bb3
    Capabilities: write_request write_command
  Characteristic: 49535343-aca3-481c-91ec-d85e28a60318
    Capabilities: write_request notify
    Descriptor: 00002902-0000-1000-8000-00805f9b34fb

D:\SimpleBLE-0.6.0\build_simpleble_examples\bin\Release>example_connect
[INFO] SimpleBLE: D:\SimpleBLE-0.6.0\simpleble\src\backends\windows\Utils.cpp:33 in initialize_winrt: CoGetApartmentType: cotype=-1, qualifier=0, result=800401F0
[INFO] SimpleBLE: D:\SimpleBLE-0.6.0\simpleble\src\backends\windows\Utils.cpp:41 in initialize_winrt: RoInitialize: result=0
Using adapter: ACERASPIRE - Top [58:96:1d:d1:53:16]
Scan started.
Scan stopped.
The following devices were found:
[0] DEVICE YYYYY [fe:24:c3:56:4a:7d]
Please select a device to connect to (0-0): 0
Connecting to DEVICE YYYYY [fe:24:c3:56:4a:7d]
Successfully connected.
MTU: 244
Service: 00001800-0000-1000-8000-00805f9b34fb
  Characteristic: 00002a00-0000-1000-8000-00805f9b34fb
    Capabilities: read write_request
  Characteristic: 00002a01-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a04-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002aa6-0000-1000-8000-00805f9b34fb
    Capabilities: read
Service: 00001801-0000-1000-8000-00805f9b34fb
Service: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
  Characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
    Capabilities: write_request write_command
  Characteristic: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
    Capabilities: notify
    Descriptor: 00002902-0000-1000-8000-00805f9b34fb

D:\SimpleBLE-0.6.0\build_simpleble_examples\bin\Release>
```

But in Linux:
```
graham@graham-dell:~/SimpleBLE-0.6.0/build_simpleble_examples/bin$ ./example_connect
Using adapter: hci0 [00:1A:7D:DA:71:15]
Scan started.
Scan stopped.
The following devices were found:
[0] DEVICE XXXXX [34:81:F4:EB:35:53]
Please select a device to connect to (0-0): 0
Connecting to DEVICE XXXXX [34:81:F4:EB:35:53]
Successfully connected.
MTU: 0
graham@graham-dell:~/SimpleBLE-0.6.0/build_simpleble_examples/bin$ ./example_connect
Using adapter: hci0 [00:1A:7D:DA:71:15]
Scan started.
Scan stopped.
The following devices were found:
[0] DEVICE YYYYY [FE:24:C3:56:4A:7D]
Please select a device to connect to (0-0): 0
Connecting to DEVICE YYYYY [FE:24:C3:56:4A:7D]
Successfully connected.
MTU: 244
Service: 00001801-0000-1000-8000-00805f9b34fb
Service: 6e400001-b5a3-f393-e0a9-e50e24dcca9e
  Characteristic: 6e400002-b5a3-f393-e0a9-e50e24dcca9e
    Capabilities: write_request write_command 
  Characteristic: 6e400003-b5a3-f393-e0a9-e50e24dcca9e
    Capabilities: notify 
    Descriptor: 00002902-0000-1000-8000-00805f9b34fb
graham@graham-dell:~/SimpleBLE-0.6.0/build_simpleble_examples/bin$
```

I don't know if the lack of services and MTU 0 on one device is the same problem as the missing service and characteristics on the other device but it looks like something isn't right.

kdewald, 2023-03-23:
Hey @GrahamNZ, let me help with this:

Some of the "missing" services and characteristics are the standard ones that contain information such as the device name and connection capabilities, which are already captured through other functions, so you won't need to use them.

A few examples include:
```
Service: 00001800-0000-1000-8000-00805f9b34fb
  Characteristic: 00002a00-0000-1000-8000-00805f9b34fb
    Capabilities: read write_request
  Characteristic: 00002a01-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a04-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002aa6-0000-1000-8000-00805f9b34fb
    Capabilities: read
Service: 00001801-0000-1000-8000-00805f9b34fb
``` 
or
```
Service: 00001800-0000-1000-8000-00805f9b34fb
  Characteristic: 00002a00-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a01-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a04-0000-1000-8000-00805f9b34fb
    Capabilities: read
Service: 0000180a-0000-1000-8000-00805f9b34fb
  Characteristic: 00002a23-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a24-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a25-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a26-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a27-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a28-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a29-0000-1000-8000-00805f9b34fb
    Capabilities: read
  Characteristic: 00002a2a-0000-1000-8000-00805f9b34fb
    Capabilities: read
```

kdewald, 2023-03-23:
Looking into this event now:

```
graham@graham-dell:~/SimpleBLE-0.6.0/build_simpleble_examples/bin$ ./example_connect
Using adapter: hci0 [00:1A:7D:DA:71:15]
Scan started.
Scan stopped.
The following devices were found:
[0] DEVICE XXXXX [34:81:F4:EB:35:53]
Please select a device to connect to (0-0): 0
Connecting to DEVICE XXXXX [34:81:F4:EB:35:53]
Successfully connected.
MTU: 0
```

I think that for some reason the connection is failing to be established and the error isn't propagated up correctly. Does this happen consistently with this device? Does it maybe require some sort of pairing?

GrahamNZ, 2023-03-23:
No problem with the missing services if there is a reason why those specific ones are omitted. I was just worried it might be something more general.

For the MTU 0 device, yes it happens all the time. It works properly with no problem every time in Windows and never works in Linux. It is paired and flagged as trusted (as it allows classic Bluetooth as well) but if I try unpairing it and watch the Bluetooth Manager screen it displays the device again when it connects. I also get the "Connected" notification so anything going wrong seems to be after that. The is_paired function shows the expected status each time.
I amended my test program so that after it finds there are no services it then waits for a keyboard entry before continuing to the Disconnect statement. What is interesting is that the connection is closing automatically (and I get the "Disconnected" notification) before I press a key, so I think you are right about there being an issue with the connection.

kdewald, 2023-03-23:
Are you using the same machine for Windows and Linux? If so, what's probably happening is the following:
- Your adapter is paired on Windows, with credentials X.
- On Linux, you try to connect with the same adapter, your device expects credentials X but the Linux OS doesn't have them.
- The device disconnects as it wasn't able to pair again.

If you have a way of triggering pairing again on your device the credentials should be regenerated.

GrahamNZ, 2023-03-23:
That would make sense, but no - different machines. So I powered off my Windows PC, turned of BT on my phone (it's paired there too), powered up the Linux machine and turned on the device. Still the same problem. It still connects OK with BT from the Linux machine though (which it does via rfcomm0).

GrahamNZ, 2023-03-25:
Good idea! I ran sudo btmon and then the example_connect program. To rule out a specific device hardware issue I used a different device (same manufacturer, same GATT, different model so different MAC address) but it had the same MTU 0 problem. I can't see anything wrong but it probably means more to you than me.

[BTmon.txt](https://github.com/OpenBluetoothToolbox/SimpleBLE/files/11067807/BTmon.txt)

A-Emile, 2024-02-10:
Hey, I have the same problem with one of my BLE devices. Did you found a solution?

GrahamNZ, 2024-02-10:
No, sorry. I suspect it's where a device has both classic BT and also BLE connection options. In my case I was able to force the user to connect using only BT for that device since the BT connection exposes a virtual com port and I can use serial comms instead. I never did get BLE working with it.

As far as I can see, Kevin's last post to the Discord chat was on 12 Dec 2023 where he explained he had higher priorities now and limited time for this project. So I'd say if the SimpleBLE version you now have doesn't do what you want then it might never do it.