simpleble/simpleble #224: [MacOS] Initial pairing attempts fail with service discovery error

BlissChapman, 2023-03-15:
Steps to reproduce:
1. Open a Macbook.
1. Attempt to connect to a device that requires the user to hit the macOS popup to connect.
2. If you don't hit the popup immediately, the service discovery process will time out in SimpleBLE and the connection will fail.
3. Subsequently trying to connect will succeed because there is no popup that needs user action so the timing of the code is not thrown off.

eriklins, 2023-03-17:
Would this be considered a bug? Pairing is done on the OS level and fully independent of SimpleBLE and hence pairing would need to be done prior to (attempting) to connect to the device. If not yet paired, it would be up to the device what to allow/expose to the client and if e.g., no services would be exposed then the above behaviour would be the result and actually not a bug.

kdewald, 2023-03-17:
It is, the connection timeouts are too strict (also on Windows I think), causing an exception to be thrown too early in the process, while the user would still have time to finish it manually.

kdewald, 2023-03-23:
Fixed in https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/233

closed 2023-03-23 by kdewald