simpleble/simpleble #231: [Python] Ensure feature parity with Cpp version

kdewald, 2023-03-19:
Some of the the newer functions in Services, Characteristics and Descriptors haven't been ported to Python.

eriklins, 2023-09-07:
Is this issue still pending? What are the "newer functions" missing in the Python version?

RyanNorge, 2023-12-08:
Not sure if @kdewald was seeing the same thing I am, but the following methods don't work for me in Python...
characteristic.capabilities()
characteristic.can_read()
characteristic.can_write_request()
characteristic.can_write_command()
characteristic.can_notify()
characteristic.can_indicate()

I get an AttributeError along the lines of "object has no attribute 'can_read'"

Also, this could very well be my own device, but service.data() is giving me empty bytes...
service.data() -> b''

Running on MacOS, so I don't know if it's a Python bindings thing or a Mac backend thing. Maybe someone can try these Python methods on a different OS?

First post on GitHub! I'm new to this computer programy stuff and I'm loving it :)

JoeHowse, 2024-02-08:
The Python methods listed in @RyanNorge's comment were missing from version 0.6.1 but are present in version 0.6.2.dev2 (currently the latest pre-release package available via Pip). To install the latest pre-release package, run `pip install simplepyble --pre`.

I suggest that this issue can be closed.

RyanNorge, 2024-02-13:
Sounds like it's probably all taken care of then. Thanks that's fantastic. It's been a busy week and I haven't been able to check the pre-release like you mentioned. Go ahead and close the issue and I'll just open a new one if I get a chance to check it and it doesn't work.