simpleble/simpleble #236: SIGSEGV crash on macOS with characteristic notify

thegecko, 2023-03-27:
While developing SimpleBLE as the new backend to the JavaScript [webbluetooth library](https://github.com/thegecko/webbluetooth), I came across a SIGSEGV fault when implementing notify support.

I checked this with the notify example in `simplepyble` and see the same issue.

Version: 0.6.1
OS: MacOS 12.3.1
Arch: Arm64
Python: 3.10.4

Test device: BBC micro:bit
Test characteristic: button - e95d9882-251d-470a-a062-fa1922dfa9a8 e95dda90-251d-470a-a062-fa1922dfa9a8

Crash dump as follows:

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               Python [98462]
Path:                  /Library/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               3.10.4 (3.10.4)
Code Type:             ARM-64 (Native)
Parent Process:        zsh [98126]
Responsible:           Terminal [58197]
User ID:               501

Date/Time:             2023-03-27 10:16:49.9786 +0100
OS Version:            macOS 12.3.1 (21E258)
Report Version:        12
Anonymous UUID:        E2B9BB5B-C613-8EAE-AFA0-62BBCBAC8D0F

Sleep/Wake UUID:       283C6B20-E253-45B6-BDE0-05B330E8F674

Time Awake Since Boot: 3200000 seconds
Time Since Wake:       237 seconds

System Integrity Protection: enabled

Crashed Thread:        1  Dispatch queue: AdapterBaseMacOS.centralManagerQueue

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [98462]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 4370431984
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1047f8000-1047fc000    [   16K] r-x/r-x SM=COW  .../MacOS/Python

Kernel Triage:
VM - Compressor failed a blocking pager_get

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1a7a975b8 __select + 8
1   Python                        	       0x105774828 time_sleep + 272
2   Python                        	       0x1055be99c cfunction_vectorcall_O + 100
3   Python                        	       0x10568cfd0 call_function + 132
4   Python                        	       0x105685370 _PyEval_EvalFrameDefault + 23864
5   Python                        	       0x10567ddb0 _PyEval_Vector + 360
6   Python                        	       0x1056e8654 pyrun_file + 308
7   Python                        	       0x1056e7d98 _PyRun_SimpleFileObject + 336
8   Python                        	       0x1056e73e4 _PyRun_AnyFileObject + 216
9   Python                        	       0x105713620 pymain_run_file_obj + 180
[... 251 lines cut ...]
      },
      "deploymentId" : 240000021
    }
  ],
  "experiments" : [

  ]
}
}

Model: MacBookAir10,1, BootROM 7459.101.3, proc 8:4:4 processors, 8 GB, SMC 
Graphics: Apple M1, Apple M1, Built-In
Display: Color LCD, 2560 x 1600 Retina, Main, MirrorOff, Online
Memory Module: LPDDR4
AirPort: Wi-Fi, wl0: Feb  8 2022 01:44:45 version 18.60.21.0.7.8.126 FWID 01-1cdae627
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB2.1 Hub
USB Device: "BBC micro:bit CMSIS-DAP"
USB Device: USB 2.0 BILLBOARD
USB Device: USB31Bus
Thunderbolt Bus: MacBook Air, Apple Inc.
Thunderbolt Bus: MacBook Air, Apple Inc.
```

thegecko, 2023-06-01:
We seem to have resolved this, will reopen it it comes back :)

Maxung, 2023-10-05:
Hello, I'm having the exact same problem, when subscribing via the notify method to a temperature characteristic I'm getting `Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)`. Could you tell me how you solved the problem? I can subscribe to the same characteristic with my iPhone and the nRF Connect without any problems.