simpleble/simpleble #237: Missing services possibly caused by underlying WinRT issue

Scooxstad, 2023-04-06:
I'm executing a basic program that scans for a specific device, connects, and lists services and characteristics. However, there are services I know for a fact should be available despite not appearing in the peripheral's services. For the record, the services I receive are Generic Access Service (0x1800), Generic Attribute Service (0x1801), Device Information Service (0x180a) and one unspecified service (a026ee01-0a7d-4ab3-97fa-f1500f9feb8b). In addition to these I am also expecting the Fitness Machine Service (0x1826), which is not present.

When debugging the application, several exceptions are thrown and caught during the execution of `PeripheralBase::_attempt_connect()`, with a call stack trace pointing to `onecoreuap\drivers\wdm\bluetooth\user\winrt\gatt\gattdeviceservice.cpp(1325)` and `onecore\internal\sdk\inc\wil\opensource/wil/winrt.h(1269)`.  The exception itself is `80070005 Access is denied`.

[image]

Debug output
```
onecoreuap\drivers\wdm\bluetooth\user\winrt\gatt\gattdeviceservice.cpp(1325)\Windows.Devices.Bluetooth.dll!00007FFF67CF67E0: (caller: 00007FFF67CEBD62) Exception(1) tid(544) 80070005 Access is denied.
Exception thrown at 0x00007FFF9B40FE7C in Sample.exe: Microsoft C++ exception: wil::ResultException at memory location 0x0000007C7E2FD690.
Exception thrown at 0x00007FFF9B40FE7C in Sample.exe: Microsoft C++ exception: [rethrow] at memory location 0x0000000000000000.
onecore\internal\sdk\inc\wil\opensource/wil/winrt.h(1269)\Windows.Devices.Bluetooth.dll!00007FFF67DB7C9C: (caller: 00007FFF67CF31A3) ReturnHr(1) tid(544) 80070005 Access is denied.
    Msg:[onecoreuap\drivers\wdm\bluetooth\user\winrt\gatt\gattdeviceservice.cpp(1325)\Windows.Devices.Bluetooth.dll!00007FFF67CF67E0: (caller: 00007FFF67CEBD62) Exception(1) tid(544) 80070005 Access is denied.
]
```

While troubleshooting this, I came across the following Stackoverflow post with two answers describing possible workarounds for a similar issue.
https://stackoverflow.com/q/71620883

Scooxstad, 2023-04-07:
After further investigation, I've determined that Bond Management Service (0x181e) is causing the Access Denied exception when attempting to fetch its characteristics. I've worked around this issue by skipping the service entirely when iterating gatt_services in PeripheralBase.cpp, which is a sufficiently valid solution for my purposes with this library.

```
if (guid_to_uuid(service.Uuid()) == "0000181e-0000-1000-8000-00805f9b34fb") {
    continue;
}
```

kdewald, 2023-04-07:
Excellent find, thank you very much! I'll fix this in the next release!