simpleble/simpleble #238: [python] [osx] RuntimeError: Not able to connect to Peripheral

serlecu, 2023-04-06:
This is the error I'm getting:

[ERROR] SimpleBLE: /Users/runner/work/SimpleBLE/SimpleBLE/simpleble/src/backends/macos/AdapterBase.mm:152 in delegate_did_discover_peripheral: Exception during callback: RuntimeError: Operation Failed: Peripheral Connection

The matched line of code in my script is a simple 'peripheral.connect()' which is proved connectable with 'is_connectable()'
The device itself is an Arduino BLE advertising a custom service.

I have no clue about the AdapterBase.mm file that is mentioned in the report.

Any insights on how to make a better report are welcome :)

serlecu, 2023-04-06:
Actually even the error spots the system shows the device as connected.

closed 2023-04-07 by serlecu

kdewald, 2023-04-07:
Hey @serlecu could you provide some more information of what the problem was and how you solved it?

serlecu, 2023-04-07:
Sorry. It was actually an error caused by wrong coding of callback function.

So, when I was implementing this function of Peripheral `void set_callback_on_connected(std::function<void()> on_connected)` I was kind of asking the function to pass a value that it wasn't returning:
`peripheral.set_callback_on_scan_found(lambda peripheral: doSomething(peripheral)` <- this  "lambda **peripheral**:"

I got confused with this other of Adapter `void set_callback_on_scan_found(std::function<void([Peripheral])> on_scan_found)` implemented like this:
`adapter.set_callback_on_scan_found(lambda peripheral: doSomething(peripheral)`. 

It happens when copy-pasting and rushing, but also bc I'm quite self-taught on programming. 😅