simpleble/simpleble #249: Scanning on Linux does not return all peripherals

ferbarra, 2023-07-04:
There are some peripherals that are only returned when performing a scan_get_results() from windows. Calling scan_get_results() on Linux (Fedora) only returns some peripherals but not all. The linux machine uses an ASUS BlueTooth 4.0 USB dongle adapter.

I can also find the peripheral using the "nRF Connect" app on Android. The peripheral I'm interested in appears as a "LE only" device and has the "Advertising type" set to legacy. Here is the raw data: 0x0409464552.

Any ideas on how to pinpoint the problem?

allanvaughanjones, 2023-08-22:
@ferbarra I think I'm seeing a similar thing. Running on Raspberry Pi OS (Bullseye). The C++ scan example finds 4 devices;
_Scan started.
Found device:  [60:04:1F:86:3A:DD] -83 dBm
Found device:  [4A:9E:BD:BA:33:D0] -75 dBm
Found device: mpy-uart [D8:3A:DD:43:4F:D5] -83 dBm
Found device:  [4F:B9:D9:07:25:99] -78 dBm
Updated device: mpy-uart [D8:3A:DD:43:4F:D5] -69 dBm
Scan stopped._

Running 
$sudo hcitool lescan 
on the same machine gives;
_LE Scan ...
3A:0E:2B:E8:55:54 (unknown)
4A:9E:BD:BA:33:D0 (unknown)
04:B9:E3:23:3C:01 (unknown)
0F:98:91:1B:BF:8A (unknown)
4F:B9:D9:07:25:99 (unknown)
07:91:5C:8B:8D:1C (unknown)
EE:48:BA:E5:7A:04 h2o383E9
7B:90:D3:9D:FC:EB (unknown)
F9:2A:F3:67:51:C6 (unknown)
CE:2A:47:C0:3C:D6 (unknown)
D8:3A:DD:43:4F:D5 (unknown)
C3:00:00:02:EF:0D (unknown)_

Did you resolve the problem?

allanvaughanjones, 2023-08-23:
Just a further bit if information on this; If I modify Adapter1 (SimpleBLE/simplebluez/src/interface/Adapter1.cpp)
void Adapter1::StartDiscovery()
 {
     DiscoveryFilter myFilter;
     SetDiscoveryFilter(myFilter);
     
    auto msg = create_method_call("StartDiscovery");
    _conn->send_with_reply_and_block(msg);
}

then I get all of the peripherals shown. There doesn't seem to be a way to set the filter from the SimpleBle API.

MikeNeilson, 2023-12-25:
I was able to take what @allanvaughanjones provided and make a slightly more generic workaround for my application:

```c++
#ifdef __linux__
    auto bluez = (SimpleBluez::Adapter*)adapter.underlying();
    SimpleBluez::Adapter1::DiscoveryFilter filter;
    filter.Transport = SimpleBluez::Adapter::DiscoveryFilter::TransportType::LE;
    filter.DuplicateData = true; // I didn't actually need these two as they are the default.
    filter.Pattern = "";
    bluez->discovery_filter(filter);
#endif
```

There's probably an even better way to properly verify it's actually using Bluez instead of DBUS or something before doing that casting, but that worked for my simple application.

And it case others have similar issue: My old laptop and Raspberry Pi 3b had Bluetooth 4.1 controllers and just wouldn't return anything even after. I bought a usb 5.1 adapter and got my expected results. (My current application is "beacon" data from an Arduino Nano BLE device.)

I also had to make sure I was using both `set_callback_on_scan_updated` and `set_callback_on_scan_found`; but I suspect that's because I'm not pairing anything, just passively listening for the "manufacture data" packets.

jcelerier, 2024-07-25:
Same here. Another workaround is to send a `scan le` command to bluetoothctl and SimpleBLE will start reporting the BLE devices too.

kdewald, 2024-07-28:
Hey, thanks to all for the feedback.

Is it possible that the discovery filter is being overwritten by some other application. I'll make sure this gets added.

kdewald, 2025-05-05:
I just had a moment to look into this. It seems that the most likely cause here is that we're relying on the default filter settings that the operating system is using and if those get modified we can be affected as well.

The fix is quite straightforward, so it will definitely make it into v0.10

Syutkin, 2026-09-30:
This issue appears to still be reproducible with SimpleBLE 1.2.0 on Linux.

I ran into a similar case with an Android phone running a BLE GATT/NUS peripheral. SimpleBLE discovery returned the phone using its Classic Bluetooth/identity address, and connecting to it caused BlueZ to use BR/EDR and fail with:

```text
org.bluez.Error.Failed br-connection-key-missing
```

Running:

```text
bluetoothctl
scan le
```

showed the same peripheral under a different LE address. Connecting to that address worked correctly and the GATT services were resolved.

Looking at the current Linux backend, `AdapterLinux::scan_start()` still calls:

```cpp
adapter_->discovery_start();

// TODO: Does a discovery filter need to be set?
```

without explicitly setting the BlueZ discovery transport to LE.

Since `SimpleBluez::Adapter1::DiscoveryFilter` already supports `TransportType::LE`, it looks like the Linux backend may still depend on the system/default BlueZ discovery filter.

Would it make sense to reopen this issue, or explicitly set `Transport = LE` before starting discovery?

kdewald, 2026-10-16:
We've recently made changes to Linux scanning on `main`, including setting an explicit discovery filter so devices that don't advertise as discoverable can be found ([ae1d6d6f](https://github.com/simpleble/simpleble/commit/ae1d6d6f)). Please try the latest `main` and reach back here if the problem persists so we can take another look.

Syutkin, 2026-10-18:
Thanks! The new scanning method helped: it now returns both addresses advertised by my Android phone, one for Classic Bluetooth and another for BLE.

While testing further with a BLE NUS server on that phone, I noticed something else.

Before pairing, BlueZ shows both addresses separately. After pairing, the changing BLE address disappears, and BlueZ shows the device under its permanent address. When testing without SimpleBLE, the first connection request established only Classic, while the second established BLE, after which NUS worked. I found out this is [normal BlueZ behaviour](https://github.com/bluez/bluez/blob/5.87/doc/org.bluez.Device.rst#void-connect): another connection request connects the other transport.

On the current main, the connection check can consider the Classic connection sufficient. As a result, connect() can finish while BLE is still disconnected, and calling it again may simply return immediately because the device is already considered connected.

So the scanning change helped, but it also exposed a separate issue with detecting an actual BLE connection on devices that support both Classic and BLE.

kdewald, 2026-10-19:
Thanks for testing! Good to hear the scanning side is sorted.

What you're seeing is how BlueZ handles dual-mode devices: once paired, `Connect()` brings up a single bearer and reports the device as connected whichever one it is. BlueZ has an LE-specific API for this (`org.bluez.Bearer.LE1`), but it's still experimental. We're working upstream to get it marked stable, and we'll use it in the Linux backend once it is.

I'll track this in #513, since this one was about scanning.