simpleble/simpleble #262: Need an code example that advertises bluetooth services

IHNEL, 2023-10-04:
Hello, I have searched around but couldn't find any example with SimpleBLE or SimpleBluez that demonstrates how to advertise new Bluetooth services that other bluetooth devices (like iphone) can scan and connect to.

My purpose is that implementing an application runs on Linux that provide some interesting features. Once the app's running, other mobile phone can connect to the app via Bluetooth and communicate to send data. 
I can make the same application on Android with Android Bluetooth APIs. On Linux, the bluetooth API is very complicated to use in C/C++ so I hope I can find an easier way using  SimpleBLE.

Thanks in advance.

eriklins, 2023-10-04:
SimpleBLE does not support BLE peripheral role and hence there is no support for BLE advertising or accepting connections from other devices. Not sure if it is planned to add support for this.

kdewald, 2024-07-28:
This feature is currently in the works, although there's no clear timeline yet.