simpleble/simpleble #265: Java bindings

KlemenDEV, 2023-10-09:
It would be neat to have Java bindings for this library. There is currently no BLE desktop Java library to my knowledge.

JNA/SWIG/JNR-FFI could be used to achieve this.

Originally this was a comment at #121 but moved to a separate ticket.

KlemenDEV, 2023-12-18:
I have started to work on Java bindings, but am currently stuck at #267

enkatsu, 2024-01-07:
@KlemenDEV Just like you, I am also developing Java bindings.
If you don’t mind, I would like to contribute to your development work. Could you please share with me the URL of your repository?

Gems, 2024-02-08:
Hello guys, I'm willing to contribute too. Would it make sense to coordinate the efforts?

lebenski, 2024-02-22:
I would love this!  Any work in progress that I can take a look at?

L4m4L, 2024-03-05:
@KlemenDEV how did you go with Java bindings? I have made good progress, but am currently stuck on passing my user-defined structs to the `simpleble_peripheral_write_command` function. Any assistance would be appreciated!

KlemenDEV, 2024-03-05:
The project got a bit stuck because it is currently not the priority at the company I was working on those bindings for

L4m4L, 2024-03-05:
No worries, I will post up my repo so others can critique it - it's close I think but I'm not implementing the Pointers correctly I think with JNI-FFI (for function calls using structs)

L4m4L, 2024-03-05:
[Here's](https://github.com/L4m4L/simpleBLEJava/blob/main/simpleBLE.java) where I have gotten up to, very messy at the moment but not sure if it is similar to where you left off?

If I can get something to work I will definitely update this code and share it with SimpleBLE

L4m4L, 2024-04-01:
Hi @KlemenDEV, I finally got Java bindings to work using Java Native Access. The weirdest thing was that I needed to change the parameter order for "simpleble_peripheral_write_command" in order for it to work...? So I think JNA was mixing up calling conventions somewhere along the way, so it may not be that stable. 

Anyway, the code is [here](https://github.com/L4m4L/simpleBLEJava/tree/main). Happy for questions/comments/feedback!