simpleble/simpleble #269: Manufacturer data is captured then removed when scanning

RomaniukRobert, 2023-10-31:
Hello. I have discovered that while scanning first manufacturer data is parsed and stored in the map PeripheralBase::manufacturer_data_ but then in update_advertising_data it is ovewritten with empty data. I suggest to apply fix: 

void PeripheralBase::update_advertising_data(advertising_data_t advertising_data) {
 .....
   manufacturer_data_.insert(advertising_data.manufacturer_data.begin(), advertising_data.manufacturer_data.end());
...
}
it helped in my case

KlemenDEV, 2023-11-10:
Could this be related? https://github.com/OpenBluetoothToolbox/SimpleBLE/issues/267

kdewald, 2026-10-27:
Fixed on main in 76fa3f01 and 60fc17e3, which ship in 1.3.0. Manufacturer data now reflects the latest advertisement combined with the latest scan response, so a scan response without manufacturer data no longer erases it.

closed 2026-10-27 by kdewald