simpleble/simpleble #276: How to change the MTU of an established connection?

spikeyamk, 2023-11-22:
I'm using an ESP32 and SimpleBLE and can't seem to change the MTU value which is by default set to 23 with NimBLE and I'm sure the client should be able to request the change of the MTU of an established connection somehow, but I can't seem to find where or how?

I'm actually able to write_request() to a peripheral with 23 bytes long but only receive 20 bytes long packets from notify characteristics from the server.

Please help, thanks in advance!

eriklins, 2023-11-23:
MTU size of 23 is the size any BLE device compliant with BT 4.x upward must support, hence it's the default (or rather min) being used upon engaging a connection. Subsequently the peripheral can initiate a connection parameter update request with a higher MTU size. However, the actual max MTU size depends on the specific implementation on either side of the connection. Also note that a larger MTU size is considered an optional feature with BT4.x upward and not all devices will support/confirm a higher value.

As far as I know it's not possible to set/request a specific MTU size with SimpleBLE (yet). It would be down to what the peripheral would request and what the underlying OS BT stack would confirm.

MTU size is the maximum size of an ATT packet but when sending a write/read/notification packet the packet always includes an OP-Code of 1 byte plus the attribute handle of 2 bytes and hence the actual data of a paket is always 3 bytes less than the MTU size.

kdewald, 2024-07-28:
Indeed, as @eriklins mentions, none of the operating systems offers an API that allows this to be changed form the host side, so this always needs to be provided by the peripheral.

closed 2024-07-28 by kdewald