simpleble/simpleble #279: this is no manufacturer data in peripheral

shashenli, 2023-12-05:
i want to capture airdrop(Apple's tech) BLE broadcast , there should be manufacturer data but i cant get it when i use Peripheral::manufacturer_data(). this is a bug

shashenli, 2023-12-05:
@kdewald @eriklins

eriklins, 2023-12-05:
Maybe check existing open issues which mention manufacturer data (there are some) and check whether your issue could be related to one of these.

I do not have issues receiving Apple BLE advertisements and can see lots of Apple devices around along with manufacturer specific data, see screenshot (https://github.com/eriklins/InsideBlue-BLE-Tool).

[image]

eriklins, 2023-12-05:
Yes, see project derscription.

shashenli, 2023-12-05:
> Yes, see project derscription.
it definitely is a bug, i cant get manufacturer data when i use Adapter::scan_for();
so i  tried anohter funcion Adapter::start_scan(),in this case i can get manufacturer data.

closed 2023-12-05 by kdewald