simpleble/simpleble #290: [Python] Application error exit with multiple devices / subscriptions

bradmartin333, 2024-02-12:
Stuck battling a multi-device application crash. Found a fix, but it requires running 2 instances of the same application. Not sure where the issue lies. Let me know if I am being too vague / more info needed.

## Pseudocode 

`main` starts new thread with a target that performs something like:
``` py
adapter = get_one_valid_adapter()
adapter.scan_for(5000)
peripherals = adapter.scan_get_results()
# device_a
device_a, service_uuid_a, characteristic_uuid_a = find_connect_device_a(peripherals)
device_a.notify(service_uuid_a, characteristic_uuid_a, lambda data: print(f"Notification: {data}"))
# Repeat for a hypothetical device_b
```

## Issue 

- device_a's characteristic notifies at 250Hz and device_b notifies at 1Hz. The data sometimes seems to get jumbled and the python program abruptly exists with no exceptions whatsoever across all main and dummy threads.
   - By jumbled, I mean I will start getting crc16 check errors where I haven't seen any before with a single device
- The only indication that there was an error is in the Windows event logs. I have not tried on Linux, yet.

## Working solution

- Now, I run 2 separate python programs with identical code but for only one device and that works
- I have a single Bluetooth 5.0 adapter on Windows

## Questions

- Is this the intended behavior?
- If I instead had an outline like:

   ``` py
   # device_a
   adapter = get_one_valid_adapter()
   adapter.scan_for(5000)
   peripherals = adapter.scan_get_results()
   device_a, service_uuid_a, characteristic_uuid_a = find_device_a(peripherals)
   device_a.notify(service_uuid_a, characteristic_uuid_a, lambda data: 
   print(f"Notification: {data}"))
   # Repeat for a hypothetical device_b
   ```

   would that change anything at all? I'm apprehensive to take the time to test that as 
   the longest time it has ran up until the crash is about 6 hours (Shortest about 10 
   seconds).
- Could it be an issue with spawning a thread to discover and connect?

Jah-On, 2024-06-08:
I'm also experiencing a similar issue on Windows where the whole app silently exits upon trying to process a notification.

Jah-On, 2024-06-08:
I'll try writing a C++ version to see if it's the Python FFI or something in the C++ back end.

Jah-On, 2024-07-17:
This appears to be an issue related to FFI as notifications in C++ work fine.