simpleble/simpleble #292: New dual license (a655d39) is not BSD-compatible

JoeHowse, 2024-02-20:
Your new dual license (a655d39) says that it creates a downstream obligation whereby some users will be subject to either a commercial license or GPL. Such a downstream obligation is not BSD-compatible. BSD-licensed open-source projects cannot use your library on these terms; if they did so, they would be passing BSD-incompatible obligations on to their users. (The same pitfall would apply to other permissively licensed projects, not just BSD-licensed.)

I strongly suggest that you rework your license text to avoid confusion and downstream problems. The text as-is (quoted in part below) implies that your license has some kind of BSD-compatibility, when it does not.

> For non-commercial or open-source projects, SimpleBLE is available under the BSD 3-Clause License, allowing free use and modification. However, if derivatives of these projects are used commercially, a commercial license for SimpleBLE is required by the entity selling the software containing SimpleBLE.

ChadNedzlek, 2024-02-21:
I'm disappointed in this change as well. I was trying to use this library for some new devices, but I'm not willing to potentially poison any downstream users with this license, so I can't take any additional updates/releases from this project.

kdewald, 2024-02-21:
Hey @JoeHowse, thanks for bringing this up. I did not foresee all consequences of this specific license design and it's clearly not my intention to poison downstream users of open source projects built around SimpleBLE.

The main goal of this change was to provide a mechanism to force commercial users to contribute at least a bit to the maintenance of this project, as otherwise this will turn into abandonware in no time once the underlying APIs change and the code degrades. **Unless someone comes up with an alternative to make the project financially viable without a commercial license (corporate sponsorships, for example), the long term existence of SimpleBLE cannot be guaranteed.**

@JoeHowse @ChadNedzlek what do you think about this modification to the licensing scheme?

For open-source projects that are explicitly listed on the README (that I'll personally review to make sure they are legit), those projects and their derivatives will have access to SimpleBLE under a full BSD license without any commercial triggers. All others not listed will be tied to the GPL/Commercial combo.

kdewald, 2024-02-21:
The bigger picture here is that the open source community has become plagued with financial problems of our own making. Every developer starting an open source project wants everybody to use it, so they'll go with the most permissive license they can find (usually MIT) to convince every passerby and their grandmother to test it out. It feels awesome once they manage to get the ball rolling and people build cool stuff with your library.

Fast forward four years and 4000 hours later, there are AT LEAST eight medical and industrial machinery companies and two universities using all of this work for free. Not only that, these companies expect the developer to add features requiring 100+ hours of work for free and will immediate stop engaging at the first sign of money being brought into the conversation.

Contrary to other SaaS projects where you can always build a product by selling a packaged version of the open source code, I have yet to see a successful business model being applied to a shared library of this kind. I've asked around and so far nobody was able to come up with an answer. Only very recently has financial support started to trickle into projects holding together half of the Internet, so good luck waiting in line until it's time for a project with a couple thousand users and is completely invisible to the public to get its share.

@ChadNedzlek my advice to you would be to build your project with a GPL/Commercial license mix from the very beginning, no matter how small or insignificant you think it will be. You can give commercial licenses away for free if you want, but you'll always have the option to capture some of the economic gains in the future if you end up adding significant value to others.

I am very open to negotiate terms that are favorable for all SimpleBLE commercial users, regardless of the size. My goal is not to squeeze a couple of pennies from the little guy, but to make sure I have the resources to allow companies to use this in industries where people's lives and safety might be involved.

kdewald, 2024-02-21:
@Andrey1994 I saw your reaction to the post. I'm making the updates to the license as we speak, BrainFlow will definitely have access to a BSD license from SimpleBLE. You've collaborated a lot into the project and that help won't be forgotten. Neither you or your users will have to worry about commercial clauses kicking in for stuff built with BrainFlow.

Andrey1994, 2024-02-21:
Unfortunately, it means I cannot use new versions of SimpleBLE in BrainFlow too...

I am aware of financial issues in open source and having exactly the same problems with my projects.
I've tried to solve it by applying to sponsorship programs like https://chanzuckerberg.com/ and others but with no luck,  not sure that you will be much more successful with it but you can try. Also, considered open source software foundations but my experience working with them was also bad.

Andrey1994, 2024-02-21:
Sorry, wrote the message before I saw your last comment

kdewald, 2024-02-21:
@JoeHowse @Andrey1994 @ChadNedzlek I just finished making the changes I mentioned in the following PR (plus some other stuff that also needs housekeeping): https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/293

I'll aim to land this soon (once I fix the current linting issues), so feel free to share your opinions.

Andrey1994, 2024-02-21:
thanks a lot!

ChadNedzlek, 2024-02-21:
Unfortunately, I have no interest in developing a GPL licensed product, so it seems like we just have radically different goals. I'll have to use the fork I've got until it fails, and then I'll have to look for a replacement when that time comes. I definitely don't want some confusing mixed license, because that leaves _my users_ trying to use my product in a very confusing state with regards to what they can do (and then what their users can do... all the way down the road).  Transitive users of this product that get it through someone you've given a BSD license too will have _no idea_ there is even another other license applied to it, so can't be bound to a license they've never seen, so will and have every right to use that copy in their uses.

It's those medical/university's choice to use something open source without paying, and you are under no obligation to provide whatever features they want if you want compensation they aren't willing to provide. It seems criminal irresponsible of them to do so, but that's their choice and to me, it seems unlikely this change will fix that. They'll just find a different square peg for their round hole.

I'm not opposed to an optional commercial license that comes with better support options (though that doesn't seem so much as a license as some parallel agreement).

kdewald, 2024-02-22:
Quick update: I finished super late with some work stuff, will look into this in depth tomorrow.

kdewald, 2024-03-03:
Thanks for all the feedback. I've swapped the BSD license for AGPL for the close collaborator projects while I figure out a better solution. Commercial license is being worked on as we speak.

closed 2024-03-03 by kdewald