simpleble/simpleble #299: Linux: SimpleBLE fails to find services for some peripherals

JoeHowse, 2024-04-07:
On Linux, SimpleBLE cannot find or use the services of peripherals which are based on the Feasycom FSC-BT986 Bluetooth module (https://www.feasycom.com/fsc-bt986). `Peripheral.services()` returns an empty list and `Peripheral.notify(...)` (with known-correct UUIDs) throws `RuntimeError: Service with UUID ... not found`.

This is apparently a SimpleBLE issue because both bluetoothctl and Bleak can find and use the services of these peripherals.

Currently, I am unsure of the cause or extent of the issue. SimpleBLE _does_ work on Linux for at least some peripherals based on other hardware.

System specs:

- SimpleBLE version: Tested with 0.7.3
- OS: Tested on Debian Bookworm, Debian Trixie, and Manjaro
- Bluetooth adapter: Tested with several from Intel, Broadcom, and Realtek

kdewald, 2024-08-19:
Hey @JoeHowse, I think this issue is now solved after landing the following fix https://github.com/OpenBluetoothToolbox/SimpleBLE/pull/321

Feel free to reopen if it's still failing.

JoeHowse, 2024-12-13:
I have re-tested and found that this issue is still present in release 0.8.1.  I do not have privileges to re-open the issue but can file a new issue referencing this one if you wish.

kdewald, 2024-12-28:
@JoeHowse thanks for the review. I do have better ways to check this now, so hopefully once I find some time I can take a deeper look.

I have one request that would help:

Using `dfeet`, could you share a screenshot of the entire DBus tree for that connected device? I have a suspicion that Bluez is rolling this characteristic as a different DBus interface object (similar to what happens with the battery data).

JoeHowse, 2025-01-02:
@kdewald Thanks, here is the screenshot of the tree for one of the Feasycom devices.

[image]

kdewald, 2025-01-02:
For the DD_0D device, can you show the interfaces for that object? They should look like `org.bluez.Device1` or `org.bluez.Battery1`

JoeHowse, 2025-01-03:
Sure, please find below a series of screenshots showing the interfaces and their methods.

[image]
[image]
[image]
[image]
[image]