simpleble/simpleble #306: Connecting directly to known peripheral under Python

ullix, 2024-05-11:
I do know the adapter, I have, and from my peripheral I do know both identifier and address. Is there a way do connect to it directly without having to go via `get_adapters` and `scan_get_results`?

eriklins, 2024-05-11:
Cannot comment on get_adapters, but it’s not possible to connect to a peripheral without prior scanning. A BLE advert is considered the invitation for engaging a connection. Without the central does not know about the peripheral and open time slots for connecting.

ullix, 2024-05-12:
Well, I do not really know how others are doing it, but at least in these 2 cases the address suffices (plus perhaps the relevant *UUID):
https://github.com/sercona/Owon-Multimeters/issues/4#issue-2285552174
https://github.com/sercona/Owon-Multimeters/issues/4#issuecomment-2104203436

eriklins, 2024-05-12:
Then the advert scanning is done somehow/somewhere automagically in lower code layers there. This is not a limitation of SimpleBLE but it's how BLE works and how it has been definied in the BT/BLE spec.

ullix, 2024-05-12:
When I run stand-alone code based on your **simplepyble** notify example, it runs well and fast for hours on end with no issues.

However, when I put this (rather simple) code into my "big code", it lasts for only 1 ... 100 sec before it crashes with a Segmentation fault! 

The big code uses PyQt5. Any suggestion what to look for? There is no PyQt5 stuff in the part of the simplepyble code.

closed 2025-06-21 by kdewald