simpleble/simpleble #308: Rust examples and library fail to build on Windows in Debug

ddassie-texa, 2024-05-22:
I'm trying to use the library via rust on Windows but it's not even able to build in the debug configuration, it fails with the following linker errors: 
```
Running `cargo build --bin=scan --package=simplersble-examples --message-format=json`...
   Compiling link-cplusplus v1.0.8
   Compiling simplersble v0.8.0 (C:\Users\builder\Documents\GitHub\SimpleBLE)
   Compiling cxx v1.0.89
warning: simplersble@0.8.0: Building in DEBUG mode
warning: simplersble@0.8.0: CWD: c:\Users\builder\Documents\GitHub\SimpleBLE
warning: simplersble@0.8.0: ENV: OUT_DIR - c:\Users\builder\Documents\GitHub\SimpleBLE\examples\simplersble\target\debug\build\simplersble-e3dbe95df36b8c7b\out
warning: simplersble@0.8.0: ENV: CARGO_MANIFEST_DIR - c:\Users\builder\Documents\GitHub\SimpleBLE
warning: simplersble@0.8.0: ENV: CARGO_PKG_NAME - simplersble
warning: simplersble@0.8.0: ENV: CARGO_PKG_VERSION - 0.8.0
   Compiling simplersble-examples v0.0.0 (C:\Users\builder\Documents\GitHub\SimpleBLE\examples\simplersble)
error: linking with `link.exe` failed: exit code: 1120
  |
  = note: "C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.40.33807\\bin\\HostX64\\x64\\link.exe" "/NOLOGO" "C:\\Users\\builder\\AppData\\Local\\Temp\\rustc6XjwWt\\symbols.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.19uv154hz0fcscrf.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.1lzq21dzu66eqzuq.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.1n4eznhib8zdoqph.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.1p43apl6s8six85v.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.1tiybajq4156ufrt.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.1z6t2ok5x91nfhs.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.1zxn1xh24t92yr48.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.26y6d3b79n347057.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.2k8rk1zq0iad922m.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.2tzmdgmr9dlewfg7.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.2wtfnfaiz0ywf0p3.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.31g1rr1c2l8q6lo3.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.3c3y32kw7izs4gy0.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.3kdt418bn307iuzm.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.41kcotplh46f7on4.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.455er47rg2cknq03.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.46cz4oeqs602dagl.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.48pzz8p3ny999hmg.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.4krpiwnu8ihkrzc8.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.4sye34d51o05s0a1.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.55x7y31tqqoul12u.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.5dlbx6gs7c1q3s11.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.5ewl1pu79a6tvm6i.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.bhkai98lxs0jixe.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.gmhm6k3l01jj0ph.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.iypttd0cj5k0z7p.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.o2offs85dp1smo7.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.tut3zh8610nthdp.rcgu.o" "c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\scan.43jc4mp81inccjk5.rcgu.o" "/LIBPATH:c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps" "/LIBPATH:c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\build\\simplersble-e3dbe95df36b8c7b\\out/lib" "/LIBPATH:C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.40.33807\\atlmfc\\lib\\x64" "/LIBPATH:c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\build\\simplersble-e3dbe95df36b8c7b\\out" "/LIBPATH:C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.40.33807\\atlmfc\\lib\\x64" "/LIBPATH:c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\build\\cxx-11fae02a0248f6d3\\out" "/LIBPATH:C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.40.33807\\atlmfc\\lib\\x64" "/LIBPATH:c:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\build\\link-cplusplus-ce527719f57d5c1a\\out" "/LIBPATH:C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib" "C:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\libsimplersble-0993ed85ab346d8f.rlib" "C:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\libcxx-4ce46a8f2332d649.rlib" "C:\\Users\\builder\\Documents\\GitHub\\SimpleBLE\\examples\\simplersble\\target\\debug\\deps\\liblink_cplusplus-a9e4cf0a10590676.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libstd-49e3d1aefc00cc02.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libpanic_unwind-42f4cb2417289cd4.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\librustc_demangle-99c77609a4536a8b.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libstd_detect-af9ffb3c6d3f8ec7.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libhashbrown-a6bfe0548f994b2d.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\librustc_std_workspace_alloc-61f07eb10bc24cc3.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libunwind-2fb3f9083307133a.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libcfg_if-6892c9ede6f0d6e6.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\liblibc-788771c7bce73875.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\liballoc-8a4c192e8601db8f.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\librustc_std_workspace_core-4bf403d115b018c3.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libcore-48aa4c2213e4ac50.rlib" "C:\\Users\\builder\\.rustup\\toolchains\\stable-x86_64-pc-windows-msvc\\lib\\rustlib\\x86_64-pc-windows-msvc\\lib\\libcompiler_builtins-34df001b737926cd.rlib" "kernel32.lib" "advapi32.lib" "kernel32.lib" "ntdll.lib" "userenv.lib" "ws2_3 [... cut ...]
          libsimplersble-0993ed85ab346d8f.rlib(AdapterBase.obj) : error LNK2001: unresolved external symbol __imp__CrtDbgReportW
          libsimplersble-0993ed85ab346d8f.rlib(Utils.cpp.obj) : error LNK2001: unresolved external symbol __imp__CrtDbgReportW
          c:\Users\builder\Documents\GitHub\SimpleBLE\examples\simplersble\target\debug\deps\scan.exe : fatal error LNK1120: 4 unresolved externals
          

error: aborting due to 1 previous error

error: could not compile `simplersble-examples` (bin "scan") due to 2 previous errors
```

I'm using rust version 1.78.0 and Visual Studio version 17.10.0.

kdewald, 2024-07-28:
We're going to rewrite the Rust backend from scratch, the current one needs plenty of improvements.

jacksonemiller, 2024-12-09:
Adding to signal on this one.

kdewald, 2025-05-20:
Reopening this issue to keep track of it, as after the rewrite I'm not sure it got fixed.

kdewald, 2025-05-31:
This is fixed now.

closed 2025-05-31 by kdewald