simpleble/simpleble #312: got Segmentation fault when execute peripheral.write_request using simplepytble in windows

YFJiang, 2024-06-06:
when i try to send "680313000a0000000000000000000000000a09696e7374616c6c657212063234383931305f34", i got  Segmentation fault,

here is my simplepyble version
$  pip show simplepyble
Name: simplepyble
Version: 0.7.3
Summary: The ultimate fully-fledged cross-platform BLE library, designed for simplicity and ease of use.

What else is needed? Thanks

bramd, 2024-06-10:
I also have crashes using SimplePyBLE in a Windows app. This is part of a large and complex application and based on the crashdump I can't determine what exactly is causing the crash/when it happens. So it might be related.

kdewald, 2024-08-19:
@bramd @YFJiang I've just landed some changes around GIL handling that might be related. Feel free to reopen the issue if installing from source doesn't solve the issue.

closed 2024-08-19 by kdewald