simpleble/simpleble #329: SimpleBLE::Safe::Adapter::bluetooth_enabled()  Calling it twice will cause a crash

gihugui, 2024-07-12:
This function will collapse when the second adjustment is applied to the current position.  

`		auto bluetooth_enabled = SimpleBLE::Safe::Adapter::bluetooth_enabled();
		if (!bluetooth_enabled.has_value()) {
			outMsg = "Bluetooth enabled status err";
		}
`

STACK_TEXT:  
00000080`78affab0 00007ff6`19e43039     : 00000000`00000000 000001d3`9c1c2830 00000000`00000000 00000000`00000000 : loveSockets!winrt::impl::factory_cache_entry<winrt::Windows::Devices::Radios::Radio,winrt::Windows::Devices::Radios::IRadioStatics>::call<<lambda_d6ec5b0ce89a03db82adf8d9e7c2c76f> & __ptr64>+0x52
00000080`78affb80 00007ff6`19de0c68     : 00000080`78affc38 00000000`00000018 ffffffff`ffffffff 00007ff6`19de03c0 : loveSockets!SimpleBLE::AdapterBase::bluetooth_enabled+0x29
00000080`78affbd0 00007ff6`19e05f5e     : 00000000`00000000 000001d3`9c1c2830 00000000`00000000 00000000`00000000 : loveSockets!`my_server::handle_open'::`2'::<lambda_1>::operator()+0x358
00000080`78affd50 00007ffb`8abc1bb2     : 000001d3`9c1c53d0 00000000`00000000 00000000`00000000 00000000`00000000 : loveSockets!std::thread::_Invoke<std::tuple<`my_server::handle_open'::`2'::<lambda_1> >,0>+0xe
00000080`78affd80 00007ffb`8b237374     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ucrtbase!thread_start<unsigned int (__cdecl*)(void *),1>+0x42
00000080`78affdb0 00007ffb`8d23cc91     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : KERNEL32!BaseThreadInitThunk+0x14
00000080`78affde0 00000000`00000000     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : ntdll!RtlUserThreadStart+0x21

gihugui, 2024-07-15:
Finally, the function of windows has been solved.

#include <winrt/Windows.Foundation.h>
#include <winrt/Windows.Devices.Bluetooth.Advertisement.h>
#include <winrt/Windows.Devices.Bluetooth.h>
#include <winrt/Windows.Devices.Bluetooth.GenericAttributeProfile.h>
#include <winrt/Windows.Foundation.Collections.h>
#include <winrt/Windows.Storage.Streams.h>
#include <winrt/Windows.Devices.Radios.h>

#include <iostream>
#include <unordered_set>
#include <map>
#include <functional>
#include <thread>
#include <atomic>
#include <io.h>
#include <fcntl.h>

#include <windows.h>

using namespace winrt;
using namespace winrt::Windows::Devices::Bluetooth::Advertisement;
using namespace winrt::Windows::Devices::Bluetooth;
using namespace winrt::Windows::Devices::Bluetooth::GenericAttributeProfile;
using namespace winrt::Windows::Storage::Streams;

bool BLEIsLowEnergySupported() {

	try
	{
		auto getadapter_op = Windows::Devices::Bluetooth::BluetoothAdapter::GetDefaultAsync();
		auto adapter = getadapter_op.get();
		if (adapter == nullptr) {
			std::cerr << "No Bluetooth adapter found." << std::endl;
			return false;
		}
		auto supported = adapter.IsLowEnergySupported();
		if (supported == false) {
			return false;
		}

		auto async = adapter.GetRadioAsync();
		auto radio = async.get();
		auto t = radio.State();                       // 获取电脑蓝牙状态 0未知，1打开，2关闭，3硬件关闭或禁用
		if (t != winrt::Windows::Devices::Radios::RadioState::On) {
			return false;
		}
		return  true;
	}
	catch (const winrt::hresult_error& e) {
		std::wcerr << L"Exception: " << e.message().c_str() << std::endl;
		return false;
	}
	catch (const std::exception& e) {
		std::cerr << "Standard exception: " << e.what() << std::endl;
		return false;
	}
	catch (...) {
		std::cerr << "Unknown exception occurred." << std::endl;
		return false;
	}
}

kdewald, 2024-07-16:
Hey @gihugui, thanks for the report.

I'll leave it open so that I'll remember to go over it soon.