simpleble/simpleble #344: Android build errors

petabyt, 2024-09-19:
```
C:/Users/brikb/AndroidStudioProjects/fudge/third_party/SimpleBLE/simpleble/src/backends/android/PeripheralBase.cpp:114:12: error: no matching constructor for initialization of 'SimpleBLE::ByteArray' (aka 'kvn::bytearray')
C:/Users/brikb/AndroidStudioProjects/fudge/third_party/SimpleBLE/simpleble/../external/include\external/kvn_bytearray.h:37:5: note: candidate constructor not viable: no known conversion from 'std::vector<unsigned char>::iterator' (aka '__wrap_iter<unsigned char *>') to 'const uint8_t *' (aka 'const unsigned char *') for 1st argument
C:/Users/brikb/AndroidStudioProjects/fudge/third_party/SimpleBLE/simpleble/../external/include\external/kvn_bytearray.h:50:5: note: candidate constructor not viable: no known conversion from 'std::vector<unsigned char>::iterator' (aka '__wrap_iter<unsigned char *>') to 'const char *' for 1st argument
C:/Users/brikb/AndroidStudioProjects/fudge/third_party/SimpleBLE/simpleble/../external/include\external/kvn_bytearray.h:30:5: note: candidate constructor not viable: requires single argument 'vec', but 2 arguments were provided
C:/Users/brikb/AndroidStudioProjects/fudge/third_party/SimpleBLE/simpleble/../external/include\external/kvn_bytearray.h:43:5: note: candidate constructor not viable: requires single argument 'byteArr', but 2 arguments were provided
C:/Users/brikb/AndroidStudioProjects/fudge/third_party/SimpleBLE/simpleble/../external/include\external/kvn_bytearray.h:56:5: note: candidate constructor not viable: requires single argument 'byteArr', but 2 arguments were provided
C:/Users/brikb/AndroidStudioProjects/fudge/third_party/SimpleBLE/simpleble/../external/include\external/kvn_bytearray.h:19:7: note: candidate constructor (the implicit copy constructor) not viable: requires 1 argument, but 2 were provided
C:/Users/brikb/AndroidStudioProjects/fudge/third_party/SimpleBLE/simpleble/../external/include\external/kvn_bytearray.h:19:7: note: candidate constructor (the implicit move constructor) not viable: requires 1 argument, but 2 were provided
C:/Users/brikb/AndroidStudioProjects/fudge/third_party/SimpleBLE/simpleble/../external/include\external/kvn_bytearray.h:24:5: note: candidate constructor not viable: requires 0 arguments, but 2 were provided
```

I added an extra constructor: `bytearray(std::vector<unsigned char>::iterator ptr, std::vector<unsigned char>::iterator size) : data_(ptr, size) {}`

And... I don't have `nativehelper`. I don't know where/how you got it for the NDK.

felixdollack, 2024-11-15:
Same here...

`ld: error: unable to find library -lnativehelper`

@kdewald can you point us in the right direction?

Andrey1994, 2024-11-17:
nativehelper is available since 31, builds ok for me with ` -DANDROID_NATIVE_API_LEVEL=android-31`

kdewald, 2024-11-18:
Hey all, thanks for all the enthusiasm around Android support!

Please be patient as things are still in an early alpha stage, but don't hesitate to highlight any issues.

felixdollack, 2024-11-18:
I am actually using it natively (C-lang) through Flutter, so it is still different from using Android/Kotlin ^^;

Andrey1994, 2024-11-18:
I am also trying to use it slightly differently.
Just started and dont fully understand how  it should work.
I have a C++ codebase with existing Java binding,  and I already have JNIenv pointer in Cpp code. And it currently uses SimpleBLE for desktop. So, I dont need any methods from SimpleBLE to be available from Java, its 100% on Cpp level.

So afaiu I need a way to pass this pointer into SimpleBLE. @kdewald I dont see how to do this wo patching, do you have any ideas?

kdewald, 2025-06-21:
Now that Android is stable, I think I can answer this question.

You can compile SimpleBLE for Android if you tell CMake where the NDK is installed:

```
# These are some hardcoded variables used for my test process. You can change them to fit your needs.
ANDROID_ARCH_ABI="armeabi-v7a"
ANDROID_API=31

ANDROID_ARGS="-DCMAKE_SYSTEM_NAME=Android -DCMAKE_SYSTEM_VERSION=$ANDROID_API -DCMAKE_ANDROID_NDK=$ANDROID_NDK_HOME -DCMAKE_ANDROID_ARCH_ABI=$ANDROID_ARCH_ABI"

cmake -H$SOURCE_PATH -B $BUILD_PATH $ANDROID_ARGS
cmake --build $BUILD_PATH -j1
cmake --install $BUILD_PATH --prefix "${INSTALL_PATH}"
```

The other thing that is needed for Android to work is to include the output JAR of the `simpledroidbridge` component as part of the final Android application. This JAR includes a few classes that need to present to forward Android callbacks into C++.

closed 2025-06-21 by kdewald