simpleble/simpleble #362: simplepyble fails to install on debian 11

ullix, 2024-11-14:
I tried to install all versions of simplepyble on Linux Mint LMDE 5 (debian 11) and each failed! Tried Python 3.9.2 and Python 3.13.0!

Like for 0.8.1:

```
$ __venv1.5.1pre17/bin/pip install  simplepyble==0.8.1
Collecting simplepyble==0.8.1
  Using cached simplepyble-0.8.1.tar.gz (109 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error
  
  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [36 lines of output]
      Traceback (most recent call last):
        File "/home/ullix/gl#16/geigerlog-v1.5.1pre16/__venv1.5.1pre17/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
          ~~~~^^
        File "/home/ullix/gl#16/geigerlog-v1.5.1pre16/__venv1.5.1pre17/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ~~~~^^^^^^^^^^^^^^^^^^^^^^^^
        File "/home/ullix/gl#16/geigerlog-v1.5.1pre16/__venv1.5.1pre17/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/tmp/pip-build-env-vno8rxct/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 334, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/tmp/pip-build-env-vno8rxct/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 304, in _get_build_requires
          self.run_setup()
          ~~~~~~~~~~~~~~^^
        File "/tmp/pip-build-env-vno8rxct/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 320, in run_setup
          exec(code, locals())
          ~~~~^^^^^^^^^^^^^^^^
        File "<string>", line 80, in <module>
        File "<string>", line 57, in is_current_commit_tagged
        File "/usr/local/lib/python3.13/subprocess.py", line 554, in run
          with Popen(*popenargs, **kwargs) as process:
               ~~~~~^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/local/lib/python3.13/subprocess.py", line 1036, in __init__
          self._execute_child(args, executable, preexec_fn, close_fds,
          ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                              pass_fds, cwd, env,
                              ^^^^^^^^^^^^^^^^^^^
          ...<5 lines>...
                              gid, gids, uid, umask,
                              ^^^^^^^^^^^^^^^^^^^^^^
                              start_new_session, process_group)
                              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/usr/local/lib/python3.13/subprocess.py", line 1966, in _execute_child
          raise child_exception_type(errno_num, err_msg, err_filename)
      FileNotFoundError: [Errno 2] No such file or directory: 'git'
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.

note: This error originates from a subprocess, and is likely not a problem with pip
```

kdewald, 2024-11-14:
Hey @ullix, thanks for reaching out.

Do none of the pre-build packages work for you? https://pypi.org/project/simplepyble/0.8.1/#files

Which one would be missing?

kdewald, 2024-11-14:
Can you try installing the following dev version?

https://pypi.org/project/simplepyble/0.8.2.dev1/

ullix, 2024-11-15:
Odd result:  (running on Linux Mint LMDE5, aka debian 11)
```
$ pip install simplepyble==0.8.2.dev1
Defaulting to user installation because normal site-packages is not writeable
Collecting simplepyble==0.8.2.dev1
  Using cached simplepyble-0.8.2.dev1-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl.metadata (3.1 kB)
Using cached simplepyble-0.8.2.dev1-cp39-cp39-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (1.6 MB)
Installing collected packages: simplepyble
Successfully installed simplepyble-0.8.2.dev1
```

Install works, and simplepyble can be used in code, all is fine.

However, as I need to run this on debian 12, I need to use Python venv, and this fails to install:

```
$ __venv1.5.1pre17/bin/pip install  simplepyble==0.8.2.dev1
Collecting simplepyble==0.8.2.dev1
  Using cached simplepyble-0.8.2.dev1.tar.gz (112 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: simplepyble
  Building wheel for simplepyble (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for simplepyble (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [132 lines of output]
      /tmp/pip-build-env-u03gk0o5/overlay/lib/python3.13/site-packages/skbuild/setuptools_wrap.py:611: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
      !!
      
              ********************************************************************************
              Requirements should be satisfied by a PEP 517 installer.
              If you are using pip, you can try `pip install --use-pep517`.
              ********************************************************************************
      
      !!
        dist.fetch_build_eggs(setup_requires)
      
      
      --------------------------------------------------------------------------------
      -- Trying 'Ninja' generator
      --------------------------------
      ---------------------------
      ----------------------
      -----------------
      ------------
      -------
      --
      CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
        Compatibility with CMake < 3.10 will be removed from a future version of
        CMake.
      
        Update the VERSION argument <min> value or use a ...<max> suffix to tell
        CMake that the project does not need compatibility with older versions.
      
      Not searching for unused variables given on the command line.
      
      -- The C compiler identification is GNU 10.2.1
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- The CXX compiler identification is GNU 10.2.1
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Check for working CXX compiler: /usr/bin/c++ - skipped
      -- Detecting CXX compile features
[... 74 lines cut ...]
          env = cmkr.configure(
              cmake_args,
          ...<3 lines>...
              languages=cmake_languages,
          )
        File "/tmp/pip-build-env-u03gk0o5/overlay/lib/python3.13/site-packages/skbuild/cmaker.py", line 354, in configure
          raise SKBuildError(msg)
      
      An error occurred while configuring with CMake.
        Command:
          /tmp/pip-build-env-u03gk0o5/overlay/lib/python3.13/site-packages/cmake/data/bin/cmake /tmp/pip-install-433jdx3s/simplepyble_e157ccb680334774a83f1fe364f78f24/simplepyble -G Ninja -DCMAKE_MAKE_PROGRAM:FILEPATH=/tmp/pip-build-env-u03gk0o5/overlay/lib/python3.13/site-packages/ninja/data/bin/ninja --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX:PATH=/tmp/pip-install-433jdx3s/simplepyble_e157ccb680334774a83f1fe364f78f24/_skbuild/linux-x86_64-3.13/cmake-install/simplepyble/src/simplepyble -DPYTHON_VERSION_STRING:STRING=3.13.0 -DSKBUILD:INTERNAL=TRUE -DCMAKE_MODULE_PATH:PATH=/tmp/pip-build-env-u03gk0o5/overlay/lib/python3.13/site-packages/skbuild/resources/cmake '-DPYTHON_EXECUTABLE:PATH=/home/ullix/gl#16/geigerlog-v1.5.1pre16/__venv1.5.1pre17/bin/python' -DPYTHON_INCLUDE_DIR:PATH=/usr/local/include/python3.13 -DPYTHON_LIBRARY:PATH=/usr/local/lib/libpython3.13.a '-DPython_EXECUTABLE:PATH=/home/ullix/gl#16/geigerlog-v1.5.1pre16/__venv1.5.1pre17/bin/python' '-DPython_ROOT_DIR:PATH=/home/ullix/gl#16/geigerlog-v1.5.1pre16/__venv1.5.1pre17' -DPython_FIND_REGISTRY:STRING=NEVER -DPython_INCLUDE_DIR:PATH=/usr/local/include/python3.13 '-DPython3_EXECUTABLE:PATH=/home/ullix/gl#16/geigerlog-v1.5.1pre16/__venv1.5.1pre17/bin/python' '-DPython3_ROOT_DIR:PATH=/home/ullix/gl#16/geigerlog-v1.5.1pre16/__venv1.5.1pre17' -DPython3_FIND_REGISTRY:STRING=NEVER -DPython3_INCLUDE_DIR:PATH=/usr/local/include/python3.13 -DCMAKE_MAKE_PROGRAM:FILEPATH=/tmp/pip-build-env-u03gk0o5/overlay/lib/python3.13/site-packages/ninja/data/bin/ninja -Dpybind11_DIR=/tmp/pip-build-env-u03gk0o5/overlay/lib/python3.13/site-packages/pybind11/share/cmake/pybind11 '-DPYTHON_EXECUTABLE=/home/ullix/gl#16/geigerlog-v1.5.1pre16/__venv1.5.1pre17/bin/python' -DSIMPLEPYBLE_VERSION=0.8.2.dev1 -DCMAKE_BUILD_TYPE:STRING=Release
        Source directory:
          /tmp/pip-install-433jdx3s/simplepyble_e157ccb680334774a83f1fe364f78f24/simplepyble
        Working directory:
          /tmp/pip-install-433jdx3s/simplepyble_e157ccb680334774a83f1fe364f78f24/_skbuild/linux-x86_64-3.13/cmake-build
      Please see CMake's output for more information.
      
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for simplepyble
Failed to build simplepyble
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (simplepyble)
ullix@xps13lmde:~/gl#16/geigerlog-v1.5.1pre16$ 
```

ullix, 2024-11-15:
More problems with simplepyble==0.8.2.dev1. The connection to this device had never been a problem in the past

```
2024-11-15 12:01:19.827 DMMsetup: Searching Bluetooth adapters
2024-11-15 12:01:19.827 DMMsetup:    Found adapter:    hci0 [34:68:95:EC:ED:E0]
2024-11-15 12:01:19.828 DMMsetup:    Selected adapter: hci0 [34:68:95:EC:ED:E0]
2024-11-15 12:01:19.828 DMMsetup: Scanning for Bluetooth devices
2024-11-15 12:01:21.292 DMMsetup:    Found [A6:C0:80:E3:85:9D] BDM 
2024-11-15 12:01:24.829 DMMsetup:    Bluetooth Scan complete
2024-11-15 12:01:24.829 DMMsetup: Connecting to: BDM [A6:C0:80:E3:85:9D]  Device: 'ow18e#1'
Traceback (most recent call last):
  File "/home/ullix/gl#16/geigerlog-v1.5.1pre16/gtools/DMM_OWON.py", line 396, in <module>
    DMMsetup()
  File "/home/ullix/gl#16/geigerlog-v1.5.1pre16/gtools/DMM_OWON.py", line 276, in DMMsetup
    peripheral.connect()
RuntimeError: The requested operation has failed.
```

tlifschitz, 2024-11-19:
Hello @ullix ,

Could you run the `which git` command inside your virtual environment and share the output? It looks like the error when building from source is due to Git being inaccessible within your venv: `could not find git for clone of fmt-populate`.

Also, could you let us know the architecture of your platform? I'm trying to understand why simpleble is being built from source rather than using a pre-built wheel. Have you tried installing simplepyble==0.8.2.dev1 with Python 3.9?

On the other hand, does the connection consistently fail with this version, or is it intermittent?

Thanks!

ullix, 2024-11-19:
I am running this from a Laptop with an Intel i5-5200U (X86_64, 64 bit). The OS had been Linux Mint LMDE5 (debian 11). Python used had been both 3.9 and 3.13 (see initial post) Pip-Installing simplepyble had failed on both.

In the meantime the OS had been upgraded to Linux Mint LMDE6 (debian 12), with a default Python 3.11.2. I can now successfully pip-install simplepyble in version 0.8.1. However, something else fails now (see later)

`which git` comes back empty, so is not installed (and had NOT been deinstalled after OS upgrade).

But when I now run the code, simplepyble finds the adapter, scans for devices successfully, even connects (I see the messages from the OS), but then fails:

```
$ __venv1.5.1pre17/bin/python gtools/DMM_OWON.py 

2024-11-19 10:23:07.581 DMMsetup: Searching Bluetooth adapters
2024-11-19 10:23:07.581 DMMsetup:    Found adapter:    hci0 [34:68:95:EC:ED:E0]
2024-11-19 10:23:07.582 DMMsetup:    Selected adapter: hci0 [34:68:95:EC:ED:E0]
2024-11-19 10:23:07.700 DMMsetup: Scanning for Bluetooth devices
2024-11-19 10:23:12.988 DMMsetup:    Found [A6:C0:80:E3:85:9D] BDM 
2024-11-19 10:23:17.703 DMMsetup:    Bluetooth Scan complete
2024-11-19 10:23:17.704 DMMsetup: Connecting to: BDM [A6:C0:80:E3:85:9D]  Device: 'select'
Traceback (most recent call last):
  File "/home/ullix/gl#16/geigerlog-v1.5.1pre16/gtools/DMM_OWON.py", line 410, in <module>
    if main():  setup_response = DMMsetup()
                                 ^^^^^^^^^^
  File "/home/ullix/gl#16/geigerlog-v1.5.1pre16/gtools/DMM_OWON.py", line 290, in DMMsetup
    peripheral.connect()
RuntimeError: The requested operation has failed.
```

No more output. The failing is in the 2nd line:
```
print(defname, f"Connecting to: {peripheral.identifier()} [{peripheral.address()}]  Device: '{owondevice}'")
peripheral.connect()
```

The device is a [DMM 'OWON OW18E'](https://www.amazon.de/gp/product/B08NJT38SF).

kdewald, 2025-06-21:
Closing this one as stale, lots of things have changed and I assume some of the underlying issues have been fixed.

closed 2025-06-21 by kdewald