simpleble/simpleble #363: How can I test for an enabled Bluetooth adapter?

ullix, 2024-11-15:
With an adapter present but no enabled, I get :
```
DMMsetup: Searching Bluetooth adapters:
DMMsetup:    Found: hci0 [34:68:95:EC:ED:E0]
DMMsetup: Selected: hci0 [34:68:95:EC:ED:E0]
[WARNING] SimpleBLE: /project/simpleble/src/frontends/base/Adapter.cpp:64 in scan_for: Bluetooth is not enabled.
```

At this stage I'd like test for enablement of the adapter (and stop further execution). How can I do this?

eriklins, 2024-11-15:
There is a function for that:

C:
    SIMPLEBLE_EXPORT bool simpleble_adapter_is_bluetooth_enabled(void);

CPP:
    static bool bluetooth_enabled();

ullix, 2024-11-15:
Sorry, forgot to say I need it for Python using simplepyble

ullix, 2024-11-15:
Works! Thanks

eriklins, 2024-11-18:
That's great! Please close the issue then. Thanks.