simpleble/simpleble #364: C Callback functions still not including peripheral handle

eriklins, 2024-11-15:
Has someone used the new C callback functions with the peripheral handle exposed as first parameter into the callbacks?

I compiled the latest commit from github (says 0.8.2 from simpleble_get_version()) on Mac OS into a .dylib. When using simpleble_peripheral_notify() I do still receive the service uuid as 1st parameter and characteristic uuid as 2nd, but 1st should be the handle now and so on.

I use the C version, not CPP.

Do I need to checkout a specific commit for that? I had assumed it should be available with 0.8.1 latest.

eriklins, 2024-11-15:
However, the C examples are working properly, so issue might be with my code...

kdewald, 2025-05-31:
They @eriklins!

The most likely explanation here is that your application is loading an older version of the dylib. Function declarations are the same regardless of building for static or shared targets.

closed 2025-05-31 by kdewald