simpleble/simpleble #367: Can I force disconnection of an "old" connection in Python?

ullix, 2024-11-23:
My code works, but every now and then there is a crash of my code e.g. due to an unexpected response of the Bluetooth device, not foreseen in my  code. 

This usually leaves the BT device connected, and in the next BT-scan I can no longer find the device. And then obiously also not connect again. While I know how to manually "remove" the BT device - annoying but working -  a user of my code may not have that knowledge. 

I wonder if it is possible to force disconnection of that "old" connection via simplepyble? Like in the same way as my Linux debian12 does it?

eriklins, 2024-11-23:
This might be due to the settings of specific BLE connections parameters like "Supervision Timeout". Your device won't consider the link as broken and hence internally drop the connection (and restart advertising) until the Supervision Timeout expired. This timeout can be up to 32 seconds according to the BT spec. Not sure what your device uses, it's part of the negotiation of connection parameters after a connection was engaged. To my knowledge there is no possibility to set this with SimpleBLE or from an OS perspective. Given this, try waiting min. 32 seconds and see if your device pops up in the scanning again. If so, it the sup. timeout and there is likely nothing to do about it.

ullix, 2024-11-23:
Well, my BT device ([DMM OW18E](https://www.amazon.de/Pawmate-Multimeter-Amperemeter-Multitester-Spannungsstromwiderstand/dp/B08NJT38SF)) does not seem to adhere to any 32 sec limit :-(.

I tested by removing the final `peripheral.disconnect()` in my code, exited the code, and waited. Even after 4 min the connection was still active, and a new BT scan failed to find the device.

I am on Linux Mint LMDE6 (debian12), which uses the [blueman-manager](https://github.com/blueman-project/blueman) for bluetooth handling. In it I can click on the device and select `remove` and the device is gone. A subsequent BT scan does find the device. Is it not possible to do this via simplepyble? 

Any other way to make it possible?

eriklins, 2024-11-23:
I guess if you just terminate your program without disconnecting, the PC's BT controller just keeps the connection active since no one told him to drop it. Might hence be a useful enhancement of SimpleBLE to implement something like simpleble_adapter_get_connected_peripherals_count() and simpleble_adapter_get_connected_peripjerals_handle() similar to the ...scan_get... functions. Not sure if that would be possible from what the OS makes available, maybe @kdewald can comment on it and consider this as an enhancement.

ullix, 2024-11-23:
Sure would be a welcom feature!

ullix, 2024-12-01:
I would like to emphsize the need for some new features / tools in simplepyble!

On my system (Linux Mint LMDE6 (aka debian12) and Python 3.12.5)  I can now **- most of the time but not always -** 1) connect, 2) run, 3) disconnect my Bluetooth DMM.

The crux is that sometimes I can not disconnect, which means that I also cannot connect again until I have removed the offending BT device (my DMM) manually using the Bluetooth Manager (blueman-manager 2.3.5, https://github.com/blueman-project/blueman).

As an aside, strangely, this "cannot-disconnect" seems to be happening more often with increasing successful connection time - the nicer it runs, the more difficult to stop it :-/. 

In such a cannot-disconnect case the scan for adapters and peripherals using simplepyble results in this:

```
==== Show Bluetooth Environment ======================================================
Searching Bluetooth adapters  
   Found adapter: #0    ID: 'hci0'  MAC: 40:9C:A7:05:76:54
   Found adapter: #1    ID: 'hci1'  MAC: F4:4E:FC:D8:10:75
   Search complete, no more adapters
Scanning each adapter for Bluetooth devices:
   Scanning adapter: #0   ID: 'hci0'  MAC: 40:9C:A7:05:76:54
      Scanning begin:         
      Scanning complete       
      No Bluetooth devices found
   Scanning adapter: #1   ID: 'hci1'  MAC: F4:4E:FC:D8:10:75
      Scanning begin:         
      Scanning complete       
      No Bluetooth devices found
```

So, it does **NOT** find any devices!

However, at the same time the Bluetooth Manager does see it and allows its disconnection, see highlighted line 'Disconnect BDM':

[image]

Only after I have done this can I re-connect using simplepyble.