simpleble/simpleble #380: pip install fails on MacOS Sequoia 15.2, Python 3.13.1

marcelloromani, 2025-01-02:
I have activated a virtualenv:

```
% python --version
Python 3.13.1

% pip --version
pip 24.3.1 from /Users/marcello/Downloads/geigerlog-v1.5.1pre19/__venv1.5.1pre19/lib/python3.13/site-packages/pip (python 3.13)
```

When trying to install the package, I get the following error:

```
% pip install simplepyble
```

```
Collecting simplepyble
  Using cached simplepyble-0.8.1.tar.gz (109 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      Traceback (most recent call last):
        File "/Users/marcello/Downloads/geigerlog-v1.5.1pre19/__venv1.5.1pre19/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
          ~~~~^^
        File "/Users/marcello/Downloads/geigerlog-v1.5.1pre19/__venv1.5.1pre19/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ~~~~^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/marcello/Downloads/geigerlog-v1.5.1pre19/__venv1.5.1pre19/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 118, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/private/var/folders/fw/dy_t6z151lx50p9h4bhkrb340000gn/T/pip-build-env-hosb6e09/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 334, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/fw/dy_t6z151lx50p9h4bhkrb340000gn/T/pip-build-env-hosb6e09/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 304, in _get_build_requires
          self.run_setup()
          ~~~~~~~~~~~~~~^^
        File "/private/var/folders/fw/dy_t6z151lx50p9h4bhkrb340000gn/T/pip-build-env-hosb6e09/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 320, in run_setup
          exec(code, locals())
          ~~~~^^^^^^^^^^^^^^^^
        File "<string>", line 82, in <module>
        File "<string>", line 54, in get_commits_since_version_bump
        File "<string>", line 32, in get_commit_since_hash
      RuntimeError: Failed to get hash: fatal: not a git repository (or any of the parent directories): .git

      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error

× Getting requirements to build wheel did not run successfully.
│ exit code: 1
╰─> See above for output.
```

Brew doesn't seem to have `simplepyble` in their repos:

```
% brew install simplepyble
Warning: No available formula with the name "simplepyble". Did you mean simple-tiles?
```

marcelloromani, 2025-02-08:
Reproducing on a fresh venv:

```
(venv) marcello@marcellos-Mini tmp % python --version
Python 3.13.2
```

OS: Mac OS Sequoia 15.3

```
$ python3 -m venv venv
. venv/bin/activate
pip install simplepyble==0.8.1
```
(venv) marcello@marcellos-Mini tmp % pip install simplepyble==0.8.1
Collecting simplepyble==0.8.1
  Using cached simplepyble-0.8.1.tar.gz (109 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  error: subprocess-exited-with-error

  × Getting requirements to build wheel did not run successfully.
  │ exit code: 1
  ╰─> [23 lines of output]
      Traceback (most recent call last):
        File "/Users/marcello/tmp/venv/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 389, in <module>
          main()
          ~~~~^^
        File "/Users/marcello/tmp/venv/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 373, in main
          json_out["return_val"] = hook(**hook_input["kwargs"])
                                   ~~~~^^^^^^^^^^^^^^^^^^^^^^^^
        File "/Users/marcello/tmp/venv/lib/python3.13/site-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 143, in get_requires_for_build_wheel
          return hook(config_settings)
        File "/private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-build-env-gy4m3ltn/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 334, in get_requires_for_build_wheel
          return self._get_build_requires(config_settings, requirements=[])
                 ~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "/private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-build-env-gy4m3ltn/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 304, in _get_build_requires
          self.run_setup()
          ~~~~~~~~~~~~~~^^
        File "/private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-build-env-gy4m3ltn/overlay/lib/python3.13/site-packages/setuptools/build_meta.py", line 320, in run_setup
          exec(code, locals())
          ~~~~^^^^^^^^^^^^^^^^
        File "<string>", line 82, in <module>
        File "<string>", line 54, in get_commits_since_version_bump
        File "<string>", line 32, in get_commit_since_hash
      RuntimeError: Failed to get hash: fatal: not a git repository (or any of the parent directories): .git

      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
error: subprocess-exited-with-error
```

marcelloromani, 2025-02-08:
A different failure happens with the latest version (0.9.0)

```
(venv) marcello@marcellos-Mini tmp % pip install simplepyble
Collecting simplepyble
  Using cached simplepyble-0.9.0.tar.gz (76 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: simplepyble
  Building wheel for simplepyble (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for simplepyble (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [148 lines of output]
      /private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-build-env-6506d_xs/overlay/lib/python3.13/site-packages/skbuild/setuptools_wrap.py:611: _DeprecatedInstaller: setuptools.installer and fetch_build_eggs are deprecated.
      !!

              ********************************************************************************
              Requirements should be satisfied by a PEP 517 installer.
              If you are using pip, you can try `pip install --use-pep517`.
              ********************************************************************************

      !!
        dist.fetch_build_eggs(setup_requires)

      --------------------------------------------------------------------------------
      -- Trying 'Ninja' generator
      --------------------------------
      ---------------------------
      ----------------------
      -----------------
      ------------
      -------
      --
      CMake Deprecation Warning at CMakeLists.txt:1 (cmake_minimum_required):
        Compatibility with CMake < 3.10 will be removed from a future version of
        CMake.

        Update the VERSION argument <min> value.  Or, use the <min>...<max> syntax
        to tell CMake that the project requires at least <min> but has been updated
        to work with policies introduced by <max> or earlier.

      Not searching for unused variables given on the command line.

      -- The C compiler identification is AppleClang 16.0.0.16000026
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- The CXX compiler identification is AppleClang 16.0.0.16000026
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      -- Configuring done (0.6s)
      -- Generating done (0.0s)
      -- Build files have been written to: /private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-install-eq8fzh24/simplepyble_45ec0b00c6154d37b7963cd0c2e96b29/_cmake_test_compile/build
      --
      -------
      ------------
      -----------------
      ----------------------
      ---------------------------
      --------------------------------
      -- Trying 'Ninja' generator - success
      --------------------------------------------------------------------------------
[... 73 lines cut ...]
        File "/private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-build-env-6506d_xs/overlay/lib/python3.13/site-packages/skbuild/setuptools_wrap.py", line 660, in setup
          env = cmkr.configure(
              cmake_args,
          ...<3 lines>...
              languages=cmake_languages,
          )
        File "/private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-build-env-6506d_xs/overlay/lib/python3.13/site-packages/skbuild/cmaker.py", line 354, in configure
          raise SKBuildError(msg)

      An error occurred while configuring with CMake.
        Command:
          /private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-build-env-6506d_xs/overlay/lib/python3.13/site-packages/cmake/data/bin/cmake /private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-install-eq8fzh24/simplepyble_45ec0b00c6154d37b7963cd0c2e96b29/simplepyble -G Ninja -DCMAKE_MAKE_PROGRAM:FILEPATH=ninja --no-warn-unused-cli -DCMAKE_INSTALL_PREFIX:PATH=/private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-install-eq8fzh24/simplepyble_45ec0b00c6154d37b7963cd0c2e96b29/_skbuild/macosx-10.15-arm64-3.13/cmake-install/simplepyble/src/simplepyble -DPYTHON_VERSION_STRING:STRING=3.13.2 -DSKBUILD:INTERNAL=TRUE -DCMAKE_MODULE_PATH:PATH=/private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-build-env-6506d_xs/overlay/lib/python3.13/site-packages/skbuild/resources/cmake -DPYTHON_EXECUTABLE:PATH=/Users/marcello/tmp/venv/bin/python3.13 -DPYTHON_INCLUDE_DIR:PATH=/opt/homebrew/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/include/python3.13 -DPYTHON_LIBRARY:PATH=/opt/homebrew/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/lib/libpython3.13.dylib -DPython_EXECUTABLE:PATH=/Users/marcello/tmp/venv/bin/python3.13 -DPython_ROOT_DIR:PATH=/Users/marcello/tmp/venv -DPython_FIND_REGISTRY:STRING=NEVER -DPython_INCLUDE_DIR:PATH=/opt/homebrew/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/include/python3.13 -DPython3_EXECUTABLE:PATH=/Users/marcello/tmp/venv/bin/python3.13 -DPython3_ROOT_DIR:PATH=/Users/marcello/tmp/venv -DPython3_FIND_REGISTRY:STRING=NEVER -DPython3_INCLUDE_DIR:PATH=/opt/homebrew/opt/python@3.13/Frameworks/Python.framework/Versions/3.13/include/python3.13 [... cut ...]
          /private/var/folders/p_/n0q6lt792cbf6hb7c3bckrj00000gn/T/pip-install-eq8fzh24/simplepyble_45ec0b00c6154d37b7963cd0c2e96b29/_skbuild/macosx-10.15-arm64-3.13/cmake-build
      Please see CMake's output for more information.

      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for simplepyble
Failed to build simplepyble
ERROR: Failed to build installable wheels for some pyproject.toml based projects (simplepyble)
```

kdewald, 2025-04-17:
Hey @marcelloromani, thanks for the report.

The next Python release will have wheels for Python 3.13. You can test them out here: https://pypi.org/project/simplepyble/0.9.1.dev170/

closed 2025-04-17 by kdewald