simpleble/simpleble #406: Deadlock in fork/peripheral

oscarthorn, 2025-04-30:
Hi!

I've been testing out the peripheral code in fork/peripheral. Mostly it seems to work very well. I even managed to bring it up to speed with main and surface some simple methods to simpleble::Adapter so I can handled services without having to interact with simplebluez.

I have noticed two problems though. The first is that if a central has connected to my device and I also try to connect to other devices the connection does not work. I.e. I'm trying to act as both peripheral and central at once. First connecting to a peripheral and then having a central connect to me works but after a central has connected I can never again connect to a peripheral (even if the central is not longer connected to me). Not sure if this i a bluez problem or something with simplebluez.

But the main issue I wanted to raise, so I can maybe get some pointers on what might be the cause, is that there is a deadlock in the code. If I call, for example the is_connected method of a peripheral that will go down through the stack and eventually lock _property_update_mutex (in Interface::property_refresh) and then call Interface::property_get which calls Connection::send_with_reply_and_block which wants to lock _mutex.

Meanwhile if the simplebluez stack (via run_async) gets a message with a changed property the it will first lock _mutex in Connection::static_message_handler and then _property_update_mutex in Interface::signal_property_changed causing a deadlock.

This problem does not seem to be present in main. Or I think the problem is there, since the relevant code is the same, it just does not seem to occur.

Stack for run_async:
__lll_lock_wait 0x00007c2059eb71fd
SimpleDBus::Interface::signal_property_changed Interface.cpp:141
SimpleDBus::Properties::message_handle Properties.cpp:139
SimpleDBus::Proxy::message_handle Proxy.cpp:316
<lambda>::operator()(SimpleDBus::Message &) const Proxy.cpp:50
SimpleDBus::Connection::static_message_handler Connection.cpp:217
SimpleDBus::Connection::read_write_dispatch Connection.cpp:124
SimpleBluez::Bluez::run_async Bluez.cpp:34

Stack for is_connected:
__lll_lock_wait 0x00007c2059eb71fd
SimpleDBus::Connection::send_with_reply_and_block Connection.cpp:156
SimpleDBus::Interface::property_get Interface.cpp:88
SimpleDBus::Interface::property_refresh Interface.cpp:120
SimpleBluez::Device1::Connected Device1.cpp:138
SimpleBluez::Device::connected Device.cpp:97
SimpleBLE::PeripheralLinux::is_connected PeripheralLinux.cpp:117
SimpleBLE::Peripheral::is_connected Peripheral.cpp:39
SimpleBLE::Safe::Peripheral::is_connected PeripheralSafe.cpp:79

The problem occurs both in fork/peripheral and in my version where I have brought fork/peripheral up to speed with main. It occurs regardless of I any of the peripheral mode code is used or not.

kdewald, 2025-05-01:
Hey @oscarthorn, thanks for the report!

You're 100% right here, we've been aware of this issue since we developed this early preview, which is why we haven't merged it into our main branch yet. There's a rearchitecture of quite a few components of the Linux backend that needs to happen first, which is currently next in line after our current set of improvements to the Android backend.

oscarthorn, 2025-05-05:
Thanks @kdewald, good to know it's on your radar!