simpleble/simpleble #412: get_paired_peripherals raises error [Windows]

josmithua, 2025-07-02:
Reproducible using:
simplypyble 0.10.3
Python 3.12
Windows 11

Code:

```python
import simplepyble

def main():
    adapter = simplepyble.Adapter.get_adapters()[0]
    paired_peripherals = adapter.get_paired_peripherals()
    print("Paired peripherals:")
    for peripheral in paired_peripherals:
        print(f" - {peripheral.identifier()}")

if __name__ == "__main__":
    main()
```

Exception:
```
Traceback (most recent call last):
  File "C:\Users\joshua.smith\Documents\path\simplebletest.py", line 13, in <module>  
    main()
  File "C:\Users\joshua.smith\Documents\path\simplebletest.py", line 6, in main       
    paired_peripherals = adapter.get_paired_peripherals()
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: WinRT Exception. Error code -2147024809: The provided device ID is not a valid BluetoothLEDevice object.
```

kdewald, 2025-07-03:
Thanks for the report! I'm neck deep on some company admin stuff I gotta finish (hopefully soon) and I'll check into this right away.

Just FYI, this feature is relatively new on Windows, so quirks are expected.

kdewald, 2025-07-09:
Hey @josmithua, can you repeat this test trying out the version from this commit?
https://github.com/simpleble/simpleble/commit/16d338f6dca5d56c5e1426d704e88504f09aaa95

It should appear on Pypi soon as version `0.10.4.dev4`.

The only change on that commit is to print a bit more information about the error and ignore it, at least until we have a better understand of what's going on.

The other thing I'd ask if you could test the C++ `list_paired` example and see if that one works as intended, just to rule out any effects from the Python bindings.

josmithua, 2025-09-18:
Hi @kdewald, I repeated the test with the new commit/release, but got the exact same result as above, with no extra logging...

kdewald, 2025-10-02:
Okay, I just pushed the fix that now should properly log the issue and skip any devices that can't be loaded. I need to fix one more thing for the latest Python build ot be out for testing.